skiselev/omega #1: Some small suggestions for improvement

sdsnatcher, 2019-10-17:
After checking the circuit, here goes some small suggestions for improvement. I hope it helps.

- It's highly recommended to have a good filter for the pin-21 of the VDP. This can improve the image quality and stay away from jailbar issues 
- The CXA-2075 is a newer TV-encoder that is pin-compatible with the CXA-1645 and offers much superior image quality. The required external circuitry is also simpler.
- Ideally, the pinout of the RGB connector should be the same as the Japanese MSX machines, even if it's a mini-DIN. Better yet if it was a normal DIN8 connector, so existing cables can be used. Another very good option is to use a DE-15 connector with the VGA pinout, as many projectors and scalers are designed to accept 15kHz this way.
- A pin header connector for the VDP ColorBus would be welcome for hardware tinkerers, following the [pinout shown in the MSX Turbo-R Datapack](http://ngs.no.coocan.jp/doc/img/datapack/7%25BE%25CF%2BMSX2%252B%25B2%25F3%25CF%25A9%25CE%25E3.%25BF%25DEP51.png). (Obs: CBDR should connected to the pin-1, and this is missing on this link). You can keep the pull-ups as you designed.
- Similarly, a pin-header to allow easy connection of a superimposer circuit
- It's recommended to add some sort of bus contention protection on the joystick ports, since a lot of European software tend to misconfigure the PSG via direct I/O, and this could fry it
- It's also recommended to add bus contention protection to the PPI port-B, as poorly written software might configure it for output, and then if some key is pressed, the port-B can be fried.
- It might be a good idea to add fusistors to the +5V pins of every joystick port, RGB-out and  for protection against short circuits. These didn't exist back at the MSX era, but are cheap nowadays.
- It might also be a nice idea to add fusistors for the +5V/+12V/-12V pins of the slots
- If the machine doesn't have the slot hot-insertion protection circuit, both the pins-44 and 46 of the slot should be connected to GND.
- It's advisable to have 100R series resistors on the slot signals, like Panasonic did in their MSX2+ and TR machines, to avoid ringing and protect the circuit
- All MSX machines have the keyboard pull-up resistors on the motherboard instead of the keyboard PCB. This allows the machine to work even with the keyboard disconnected, for easier maintenance.
- A pin header to connect RESET and NMI buttons will be welcome for developers

Last, some questions:
- Where does the KBD_WAIT signal comes from? I couldn't find it anywhere

skiselev, 2019-10-18:
Thank you for your suggestions!
Generally speaking I'd like to keep this design as simple as practically possible. So some features are intentionally missing. Also at this stage I don't want to make significant schematic or PCB layout changes.

- VDP pin 21 is only connected to a 470 ohm pull-down. It is not used to produce composite video. Also V9958 uses this as an additional VCC. I am not sure what filter needs to be there and why.
- CXA2075: I didn't want to use SMD components. But since CXA1645 and CXA2075 have similar layout it should be possible to make a simple SOIC to DIP adapter PCB. BTW, I was considering using AD724 but decided against it because it is an SMD IC
- I've changed both RGB and Cassette connector to DIN-8 with the standard MSX pinout. Please check the layout / schematic again.
- I'll check if I can squeeze in a color bus connector
- I am not quite sure what is a practical use of superimposer
- PSG/PPI protection - I am not sure what kind of protection? Just add resistors in series?
- Polyfuses - generally modern power supplies have short circuit protection. Also we are talking about only a few amperes current at most. With that being said, I'll see if I can add polyfuses for joystick ports. YS/AV pins of RGB are connected through resistors, so they should not cause short circuit.
- Pins 44 and 46. I'll fix it. Thanks! By the way, I did some research. I couldn't find any definite answer to where these pins should be connected. At least one MSX2 machine has only pin 44 is connected to the ground.
- 100 ohm to reduce ringing - might be a bit of overkill for 3.5 MHz bus. I also use TTL bus drivers that have somewhat slower switching characteristics compared to CMOS.
- Keyboard pull-ups. I'll fix it. Thanks!
- Pin headers for NMI and RESET - definitely a possibility. I'll see if I can squeeze in these.
- KBD_WAIT - is a leftover from earlier version. It was supposed to help with keyboard emulation using a micro-controller. It is not relevant any more and can be removed.

sdsnatcher, 2019-10-18:
- VDP pin21: The V99x8 is very susceptible to noise. Yamaha provided a separate supply for the V9958 DAC as they learned their lessons on the V9938. If you google for "Philips V9958 upgrade noise" you'll see that the most common problem of MSX2 machines upgraded to use the V9958 is a noisy video output. Nothing fancy is required, just a simple/compact LC filter will do, preferably one for each Vcc pin, and as close as possible to that pin. Given that many peripherals (that will be connected to the slot) now have switching power supplies, it's recommended to add LC filtering to the CXA-1645 Vcc pins too.

skiselev, 2019-10-19:
I've fixed the following:
1.  VDP and CXA1645 power supply now goes through a pair of LC filters.
2. Added a polyfuse for joystick ports... A single 740 mA polyfuse should be enough for both ports.
3. Connected slots pin 46 to GND
4. Added a 4.7 k pull-up resistor array for all keyboard column inputs.
5. Added a header for RESET and NMI

I've posted the updated schematic (and PCB design). Please take a look.

I am still unsure about color bus connector. I haven't found any practical use. In S-100 VDP board they've tried to connect a RAMDAC, but they've dropped the idea in the later versions. Same goes for superimposer.

sdsnatcher, 2019-10-19:
Note: please keep in mind that anything I state here will be just suggestions/brainstorming, like a friendly chat in a bar or retro meeting. By no means I want to sound that I'm trying to impose anything, ok?   :)

> I didn't want to use SMD components.

Ok, I wasn't aware of that. It's a design choice. One idea could be to overlay the pads for the two types: DIP and SMD at the same place (you just have to offset them by 50%). This way the owner could choose the one that he feels more comfortable soldering, or could buy (AFAIK, the DIP version of the CXA1645 is harder to find than the SMD CXA1645 and CXA2075)

> I've changed both RGB and Cassette connector to DIN-8 with the standard MSX pinout. Please check the layout / schematic again.

The pinout is correct. My suggestion is that it's better to send pure CSYNC (1.8Vpp) or Luma (the easier choice) to the pin-4 of the RGB connector instead of CVBS, because the color carrier of the composite-video signal is known to cause strong jailbar interference on the RGB video. (check the FS-A1GT jailbar fix at [this link](http://frs.badcoffee.info/hardware/MSXfixes.html))

> I am not quite sure what is a practical use of superimposer

The superimposer is required to play either [LaserDisc](https://en.wikipedia.org/wiki/LaserDisc) or [VHD](https://en.wikipedia.org/wiki/Video_High_Density) games. Since the majority of the MSX models don't have the imposer (and no easy connector to install one), gamers end up restricted to a choice of a few expensive/rare models.  And you don't need to buy the expensive player/discs anymore, since Louthrax created the nifty [PiLD player](https://www.youtube.com/watch?v=BSolVjAUy7M) based on a Raspberry Pi.

The imposer was also used for subtitling VHS tapes (there are some very good MSX2 softwares for that), but probably nobody will use it for that anymore nowadays.    ;)

But please notice that it's not the superimpose circuit that I'm suggesting to add. Just a pin-header so anyone who wants to add the circuit later can do it easily.

> PSG/PPI protection - I am not sure what kind of protection? Just add resistors in series?

I checked with a friend of mine who's a technician, and he mentioned that the series resistors would be a cost effective way to prevent things to be fried. You have to calculate the resistor value based on the PSG type and 74* series (LS, HCT, HC etc) you have chosen. He's wild guess is that 100R should be enough.

> I am still unsure about color bus connector. I haven't found any practical use. In S-100 VDP board they've tried to connect a RAMDAC, but they've dropped the idea in the later versions.

The colorbus is used for many things. When used for input, one of them is digitizing images in realtime. [Something](https://www.youtube.com/watch?v=8yimCUW0Dw0) [that](https://www.youtube.com/watch?v=-joo9ocbEOc) nerds seem to [love](https://www.youtube.com/watch?v=qV7Kw7mZREw).   ;D
Yes, it could have been used to implement something like the SuperFX on the MSX2/2+, but with way more bandwidth.

When used for output mode, it can indeed be used for a RAMDAC. [It was used like this on the CAPTAIN Multi Station](https://www.msx.org/wiki/NTT_Captain_Multi-Station). The Yamaha V99C37 RAMDAC was connected there.

On the input mode there's also the mouse function, but that's only available on the MSX2 and it's useless.

But it's better to keep it as simple as possible: One idea could be to just reorder the pins of the RR2 and RR3, so the whole colorbus get the far-end (higher numbered pins) of the RR2 (plus one extra hole for the CBDIR). This way, to have access to the color bus it would be enough to change the RR2 for one with less pins, and solder a pin header on the pins that go to the colorbus. What do you think? It wouldn't cost you any extra space or dollars.   :)

Same deal with the pins for the light-pen (available only on the V9938). They could be reordered to be the last 2 pins of the RR3. The MSX2 BIOS has support for this light-pen, and it was implemented on two models from Daewoo.

sdsnatcher, 2019-10-19:
- Maybe it's better to use +12V/-12V to supply the opamps. This way you won't have to worry about biases, and how it might interact with the audio circuit of the cartridges that will be connected to the slot.

skiselev, 2019-10-19:
A couple of questions:
- I think that for color bus it won't be enough to have just the color bus data and the direction signal. At least the dot clock is required for simple RAMDAC application. Also GND and 5V would be useful if. What other signals does it need?
- Superimpose - what signals we are talking about here? YS? Video signals / sync?

Comments:
- Single supply op amp LPF works just fine. The inputs are decoupled with capacitors (even in case of a dual supply op amp they are), so it should not make any difference for cartridges sound output. Nothing else on the board uses 12V/-12V. I assume some cartridges might... In this case I provide a header to connect 5V/12V/-12V supply.
- I will swap the slot numbers
- It should be possible to solder a right angle edge connector for the top slot. The only problem is to find one...

sdsnatcher, 2019-10-19:
> I think that for color bus it won't be enough to have just the color bus data and the direction signal. At least the dot clock is required for simple RAMDAC application. Also GND and 5V would be useful if. What other signals does it need?

You're right. According to the Philips NMS-8280 service manual, the dot clock (DHCLK) is also needed. 

- The superimposer is easier to be done on the V9958, so I would worry about this one. The signals required are: YS, DLCLK,  VRESET, HRESET. 

> It should be possible to solder a right angle edge connector for the top slot. The only problem is to find one...

Tip: they're being sold [here](https://www.ebay.com/itm/5pc-Industrial-Card-Edge-Slot-Socket-Connector-25x2P-50P-2-54mm-0-1-Right-Angle/130970278382?hash=item1e7e6fd9ee:g:kxAAAMXQHeBSEH~l) for many years. He also has a [listing for larger quantities](https://www.ebay.com/itm/50x-Industrial-Card-Edge-Slot-Socket-Connector-25x2P-50P-2-54mm-0-1-Right-Angle/130970278891?hash=item1e7e6fdbeb:g:kxAAAMXQHeBSEH~l) if needed. I purchased those to replace defective slots on some of my MSX machines, and they worked very well. The fact that they have no flanges wasn't a problem. (BTW, he also has the straight slot connector type if you need them)

An alternative trick is to design a male EDGE connector on the PCB facing backwards, and solder the slot connector directly on it. People have been doing this trick for years [to add the rear slot to the Yamaha CX5M computers](https://www.msxpro.com/cx5m.html#slot), and it works like a charm.

sdsnatcher, 2020-09-10:
I have a suggestion for the RAM circuitry that can save space on a next revision of the motherboard, see if you like it:

- Eliminate U32
- Add the following pins to the J12 header:
  - /RAM_SL and /RAM0_CS, side-by-side. They'll be connected by a jumper when no RAM expansion daughterboard is present
  - A19 to A21
- Eliminate the /RAM1_CS to /RAM7_CS pins from the J12 pin header. A RAM expansion board could easily generate itself /RAM0_CS to /RAM7_CS from the other pins of J12.
- Eliminate U26. If necessary, it can also be placed on the RAM expansion board.

skiselev, 2020-09-10:
I actually was considering this implementation initially. U32 is indeed optional, it is likely that most users will not ever upgrade the memory beyond 512 KiB, and in this case U32 can be omitted and /RAM_SL can be connected directly to /RAM0_CS.
I am unsure about U26... Eliminating it might make system less stable. There are too many things connected to the data bus already ;-) We can do an experiment, and replace it with some wires :-)