skiselev/omega #2: Suggestion: Turbo support

sdsnatcher, 2019-10-17:
I added this suggestion here on its own issue, to avoid cluttering on the MSX.org news thread.

In fact, the [circuit for the turbo is very simple](https://hansotten.file-hunter.com/do-it-yourself/clock-upgrades/#what-is). For the MSX1 and MSX2, the turbo will be disabled on all I/O operations, but it still can provide a very welcome boost in performance. An MSX1 at 5.37MHz with this turbo can run ZX-Spectrum ports at their correct speed, i.e.

As mentioned before, for the MSX2+ it's possible to speed-up the access to the VRAM, thanks to the V9958 /WAIT feature. For this, Belavenuto and myself have designed a GAL logic that:

1) Turns the turbo ON/OFF at the switched I/O port 41h 
2) Disables the turbo for a selection of I/O ports, but not the VRAM I/O port.

It only can't read the status of the turbo, as it was designed to complement implement the turbo on Sanyo2+ machines, and those already provide the turbo status feature internally. You probably would be able to easily add the turbo status feature

If you're interested, I can gladly send you the .EQN file for the turbo GAL as an open-source contribution.

sdsnatcher, 2019-10-18:
Temp

> frankly, the benefit won't be that great.

I beg to disagree. Once you have used a turbo MSX machine, you won’t want to go back and will start wondering why they haven’t done this before     ;)

There are plenty of heavy games and apps that will run much better when a turbo is present. Just to name a few:

- Games: Metal Gear 2, Space Manbow, King’s Valley 2, Penguin Adventure, Undeadline, Snatcher, SD Snatcher, Valis 2, Ys 3, any ZX-Spectrum port, any PC-88 port, Aleste, Aleste 2, 1942, Gunship, Sim City, Red Zone, Tales of Popolon, Playball 3, Xak1/2/3, Fray, Illusion City, etc
- Apps: Nextor/MSX-DOS2, SymbOS, Graph Saurus 2.0, Philips Video Graphics, Print Shop II, InterNestor, MUE, MGSplay, etc

> To replicate turbo functionality I would
need to implement: 
> Glitch-less clock switching circuit, an I/O port to turn on/off turbo, 

As you have seen, the SuperTurbo design is very simple and compact

> a wait state generator to generate additional wait states when accessing PSG 

SuperTurbo takes a very simple approach to this. It just lowers the clock when needed

> and perhaps the cartridge slots. 

Not really necessary, for speeds up to 10MHz. But the approach I use when I want to be conservative is to connect the SLTSL pin of the slot-1 to one of the SuperTurbo break inputs. This allows a flexible configuration where the slot-2 is the “turbo slot” and the slot-1 is the “slow slot”. On a new motherboard like this, configuration switches could be added to select if a slot will be turbo or not.

sdsnatcher, 2019-10-19:
> As far as circuit protection and turbo goes, I simply don't have the real estate on the board to implement these. Turbo is on my wish list for a CPLD implementation.

Ok.  For the current PCB, maybe it could be added as a daughterboard later. It seems to be possible since you took great care on the mobo design. Just please add simple jumpers to each slot /CPUCLK signal, to make it easier to inject the turbo clock there for a specific slot if needed.

Due to being very simplistic, the SuperTurbo is a good design to add turbo for older machines, or when space/cost/DIP-requirement are of concern. If you're going to use a CPLD, then it's much better to take the wait-state generator approach. 

The VDP requires a custom waitstate generator. Please let me know at this thread when you decide to implement it, so I can come back to describe the specifications. A generic waitstate generator will have a horrible impact on the CPU->VDP access, as happened on the MSX Turbo-R.

Pax-nl, 2025-04-17:
Is there an update for this enhancements?
There are several mentions about a Turbo “mode” and it is also state it is on the list of enhancements for future releases..

I ask because games like Blade Lords can really benefit from a speed boost. Next to the games and apps SDSnatcher mentioned.