skiselev/omega #5: Simple floppy disk controller

Wierzbowsky, 2019-10-18:
A computer without a floppy disk controller requires purchasing an additional cartridge that offers IDE or FDC functionality. I would recommend to add a simple floppy disk controller (TDC-600 or similar) onto the board and a 34-pin connector for a floppy drive or Gotek emulator. Also a power socket with 5v and 12v could be useful to connect the Molex power connector for a drive.

skiselev, 2019-10-18:
FDC will not be added in the current revision of the project.
It is already on my list of the future upgrades. It is quite possible that I'll remove printer and cassette recorder circuits (who needs that? 😄) and replace it with the FDC.
By the way, 3.5" floppy drives only need 5V power supply and do not use 12V.

Wierzbowsky, 2019-10-18:
Better keep LPT and CAS for compatibility. If disk drive dies, the only way to load games would be the cassette port. The LPT is rarely used, but it's better to keep it for the authentic feeling.

sdsnatcher, 2019-10-19:
I like the idea of an internal floppy drive, but some ponderations must be done considering the way this machine was designed.

Adding an internal floppy interface is only useful if the floppy is internal too. Otherwise, it will only be an annoyance, because:

1) It will slow down the boot when the drive isn't present. The interface will keep waiting for the drive until it times out
2) It will eat up memory for nothing
3) If someone has an external drive+power supply, what's the difference of having a cartridge interface to use it too?

But if the drive is going to be internal, this machine was designed with the two slots occupying the right side of the PCB. This means it wont be possible to install an internal floppy on the right side like all all-in-one machines like the MSX and compact Amigas have. Maybe the floppy could be placed facing the left side, but it feels so awkward.   :)

The TDC-600 has the advantage of having a PC FDC, but its address decoder circuit is stupid.  That should be improved to place the FDC at the correct range at the end of the frame-1, because the way the FDC has to be accessed now requires a lot of  unnecessary slot juggling and is incompatible with the MSX-Audio extension.

skiselev, 2019-10-19:
I am open for suggestions for modifying TDC-600 or similar FDC. I was planning to build one for my system anyway.
Relocating FDC registers elsewhere would require modifications to the floppy BIOS. Is there a source available?

sdsnatcher, 2019-10-19:
Disassembled sources for the FDC drivers of nearly all MSX interfaces are available [here](https://sourceforge.net/projects/msxsyssrc/). More specifically, at [this folder](https://sourceforge.net/p/msxsyssrc/git/ci/master/tree/diskdrvs/).

If you're going to implement a brand new floppy disk interface cartridge, some other improvements could be done that would be **very welcome**:

- Fix the TDC-600 design mistakes
- HD floppy disk support (1.2MB & 1.44MB)
- Native Nextor support

If such an interface was built, I would buy it [without blinking](https://i.kym-cdn.com/entries/icons/original/000/005/574/takemymoney.jpg).    :)

------------------------------
Here's one draft of how a new disk interface cartridge could be implemented:
(It's meant to be as close as possible to the Turbo-R disk interface design, with some enhancements):

7FF0h: DiskROM frame-1 page select
7FF1h: Digital Status register
 when read:
   bit0: /HD0
   bit1: /HD1
   bit2: /READY0
   bit3: /READY1
   bit4: /DiskChange0
   bit5: /DiskChange1
   bit7: FDC IRQ pin status
 when written:
   bit7: 0=Suppress FDC interrupts, 1=Allows FDC interrupts  (this must be ANDed to the FDC IRQ pin, and the result negated and sent as open-collector to the MSX /INT pin)

7FF2h: Operations Register		(w/o)
7FF4h: Main status register		(r/o)
7FF5h: Data register			(r/w)
7FF8h: Configuration Control Register	(w/o)

7FFEh: 1kHz countdown timer (used to wait for timeouts, regardless of the CPU speed)
7FFFh: DiskROM frame-2 page select (to simplify flashROM updates)

7C00h-7DFFh: Data transfer window	(r/w)  Available only on page-7, allows LDIR to be used for DMA-like data transfers. When accessed:
  - Must activate the /DACK pin of the FDC
  - Must generate /WAITs for the Z80 based on the FDC /DMA pin status

To allow unmodded PC drives to work easily, PIN-34 should be kept as /DC and pin-2 should be /RDY (with a pull-up resistor). Yes, this is the opposite of the Shuggart interface (and Amiga), but this way the /RDY will work just as a accelerator bonus for those who mod the drive. It will also be much simpler to mod the drive.

[image]

msx-solis, 2020-02-21:
It would be possible taking advantage from the SD & floppy circuits from the N8 project?
https://www.retrobrewcomputers.org/doku.php?id=boards:sbc:n8:n8
It would be great having a hard disk included in the mainboard.

skiselev, 2020-02-21:
Technically it is not difficult to implement floppy or CF interfaces.
For floppy there is WC37C65 FDC controller that several of my previous projects and N8 uses it too.
CF cards support a 8-bit interface that can be connected directly to Z80 (perhaps adding a 74HCT245 transceiver). But then CF cards are becoming somewhat difficult to obtain.
SD cards support is not that easy to implement. N8 uses a built-in Z180 serial port to implement it. Obviously, there is nothing like that on Omega.

As I've already mentioned, I have several issues with implementing either one of these storage interfaces:
- Software support. I don't have time to do that. If there are volunteers that can write Disk BIOS, we can talk
- Board's real estate. In order to add another component, I'll need to remove an existing one. The printer port and the cassette recorder interface are likely candidates to be removed. Of course I can go CPLD route to reduce the number of discrete logic ICs and free up some space. The disadvantage in this case is that builders will need to get a JTAG adapter to program CPLDs. Also the availability of 5V-tolerant SPLDs is questionable (there are a few that are pretty much on life support)
- Finally, cartridges that implement mass storage functionality are available. Which makes it mostly a 'nice to have' feature. Yes I get that some MSX people want to have 2 slots available for something else... but then there are also slot extenders ;-)

msx-solis, 2020-09-06:
I think we can add a pin header in future OMEGA versions, something like a third cartridge but I / O. Several people would like to have a disk system without losing any of their cartridge slots, however, I don't think everyone wants the same disk system and it is best not to impose it.
Since OMEGA's ROM is so easy to modify, my solution would be to add a pin strip that would allow you to connect an I/O cartridge port.
That way, everyone could add an interface without memory and their own disk system, modify the BIOS to their liking. Furthermore, this solution does not force to increase the size of the motherboard.
Anyone who wants a disk system could plug in a ROMless TDC-600 and add that disk BIOS to OMEGA's ROM. If you prefer a compact flash disk, do the same but using that interface. If they don't want a disk, no one will have to wait for a delay in system startup.

JmanNZ, 2020-10-04:
Just so I am clear will the TDC-600 cartridge work as is or does the bios require a modification ?

msx-solis, 2020-10-04:
What I'm suggesting is a way to connect a floppy interface (or another disk interface) without loosing a slot card on it. My suggestion was to have an internal i/o slot, not full slot cartridge.
unfortunately that kind of disk interface is not common on the MSX, they used to use memory ranges to communicate even with WD34C65 chip, not only on-cartridge BIOS expansion.
In some other Z80-CPM systems the communication with that chip was through I/O ports.

JmanNZ if you build the TDC-600 and insert it on a slot cartridge it will work without BIOS modification

Leshak-pcb, 2023-03-03:
I'm currently building my Omega MSX & I also want to make a FDC for it. My suggestion is to use a i82077 controller, at 7FF0h. Only 3 ICs needed for address decoder - 74ls133 for A4-A14, one inverter for /A3 (another one for IRQ, if needed) and 74ls139 to select either FDC or ROM, with its enable pin connected to SLTSL. All main registers (data, control, operations) will line up with turboR FDC. WD37C65, TC8566F & i82077 are uPD0765 compatible, even direct-access software should work.
Although after some digging through FDC ROM sources I've ended up with a thought that a new one is needed, made from scratch. And my programming skills are barely enough to patch existing ROMs.

msx-solis, 2023-03-03:
Hi Guys.
I have a new FDC controller design for the omega MSX based on Microsol CDX-2 FDC by port.
I have two version for this controller board, as slot cartridge and as internal cpu hat (to avoid spending a slot for the controller) of curse you don't need to upgrade BIOS for any cartridge version but you have to add this ROM driver to the omega system bios if you want the internal version.
I have just shared the cartridge version on my github and I will share the internal one soon.
regards.

dude22072, 2023-12-05:
@msx-solis Do you still plan on sharing the internal version?

msx-solis, 2023-12-06:
@dude22072 yes, but I need time enough to do it.
send me an email to have the gerbers, please.

atssada, 2024-06-15:
> @dude22072 yes, but I need time enough to do it. send me an email to have the gerbers, please.

Can I have the gerbers, i would like to have internal fdd.

atssada@gmail.com 

Best,
Kim.

Pax-nl, 2025-01-31:
@msx-solis is your internal option still email only, or already publicly available?