skiselev/omega #7: Stop button location and size

Wierzbowsky, 2019-10-18:
On most of MSX computers the Stop button is large and is located in the upper right corner of the keyboard. In your design the Stop button is indistinguishable from surrounding buttons, which is not good.

I would recommend to put the Stop button in the top right corner if you don't want to change the design of the PCB.

Also where is the Select button? I can see only Ins, Del, Home and Stop buttons.

skiselev, 2019-10-18:
For the ease of construction I use inexpensive and readily available PC 104-key keyboard keycaps. And so I won't be able to change the button size (PC keyboards simply don't have a similar large key).
I can consider moving keys around, but I am worrying that a shape of the keycaps will be an issue. At least the keycaps that I use have slightly different curvature depending on the row.

LEDs are located under Caps and Kana/Run keys respectively. Multiple options to make these LEDs visible:
1. Use "backlight" type of keycaps
2. Drill a small hole in the keycap so that the LED will be visible
3. Use brighter while, blue, or "true green" LEDs that will be visible in between the keycaps.

sdsnatcher, 2019-10-19:
The Panasonic MSX2/2+/TR machines have STOP keys at a normal size, at the top center of the keyboard. Sanyo machines have STOP keys with the same size as the other keys. Same deal for Sony.

Personally, I don't find it important to have an oversized STOP key.