skiselev/omega #14: Minor suggestion - Mouser BOM no longer accurate

billlagr, 2020-09-13:
Hi,
This is a really minor thing, but the BOM on Mouser is no longer correct - there's some obsolete parts, and also a large number of parts have had their minimum order quantity changed to 200! (eg resistors). The list may have to be updated with substitute parts.

skiselev, 2020-09-14:
Thanks for your feedback. I'll update the resistor part numbers in the BOM likely using Yageo parts (such as https://www.mouser.com/ProductDetail/603-MFR-25FRF5220K), although any other 1/4W resistors can be used. The only out of stock, obsolete part is the trimmer capacitor. It can be replaced with a regular ceramic capacitor (see BOM for the part number), or it can be purchased elsewhere.

elosha, 2021-07-28:
@skiselev
• Could you please add alternate transistors/diode types for EU parts in the BOM, like maybe PN2907A –> BC327?
• Furthermore the relais G5V-1-T90-DC5 is a high temp part that is often isn't available, while the basic G5V-1-DC5 is more common.

Thank you :)

skiselev, 2021-07-28:
@billlagr 
Apologies for a long delay. I've updated the resistors and resistor arrays part numbers to ones that are normally available in single quantities from Mouser

@elosha 
I've added the G5V-1-DC5 relay part number to the BOM. Thank you!
Regarding the transistors, PN2907A is an extremely common part, likely the most common PNP transistor. There is no longer European vs US parts... [Everything is made in Taiwan (c)](https://www.youtube.com/watch?v=bifOI4MbHVU) by ON Semiconductor... :)  BC327 has a different pinout - emitter and collector are switched. Otherwise the specs are pretty similar, so if you're inclined so, you can use these transistors

closed 2021-07-28 by skiselev