skiselev/omega #18: Carnivore2 multifunction card support

Sch-LikA, 2021-05-16:
Hi !

I am super interested in building myself an Omega computer, it looks amazing, thanks for the hard work !

I came accross the [Carnivore2](https://github.com/RBSC/Carnivore2) open-source multifunction card : CF support, RAM extension, FlashROM, multi-sound chips emulation, ...

Has anyone tested it on the Omega ? Does it work OK ?

Thanks !

skiselev, 2021-05-17:
Yes, Omega has been tested with Carnivore2 cartridge. It works great!
Note that it requires a DC-DC converter to produce +12V/-12V voltages that audio amplifier in Carnivore2 uses.

Sch-LikA, 2021-05-18:
Amazing !! Thanks for the answer. I'll definitively build one !

closed 2021-05-18 by Sch-LikA

serge-intel, 2021-05-18:
Martin Lukasek documented building +12V/-12V DC-DC converter here:
https://www.8bity.cz/2020/power-module-12v-for-omega-home-computer/
Also here is his post on using Carnivore2 with Omega Home Computer:
https://www.8bity.cz/2020/carnivore2-compact-flash-vce-pameti-a-lep-zvuk-pro-msx/

skiselev, 2021-05-18:
Martin Lukasek documented building a +12V/-12V DC-DC adapter here:
https://www.8bity.cz/2020/power-module-12v-for-omega-home-computer/
Also, here is his blog post about Carnivore2:
https://www.8bity.cz/2020/carnivore2-compact-flash-vce-pameti-a-lep-zvuk-pro-msx/

Sch-LikA, 2021-08-31:
Cool thanks for the info !