skiselev/omega #24: Power suppy issues

merlinkv, 2021-12-24:
Hi Sergey.

First of all, I'm an enthusiast of your project. Thank you very much for your work.

I have an 1.4EU board and then I have made two versions of that board myself.

The fact is that **with all of them I have the same power issue**.

For my tests I use a regulated laboratory PSU.

Without added cartridges or modules the Omega boot perfectly at +5.2v, never at +5.0v. But if I put the 4MB RAM upgrade or use cartridges like Carnivore 2, MegaFlashROM or TDC-600, the only way to get it to work properly is with 5.47v / 5.50v.

It does not matter if I use a current of 3A, 3.5A, 4A or 5A ... the result is always the same. If I want to use the RAM expansion and cartridges, the only way is with 5.47v/5.5v.

- No heated components
- All TTL are HCT
- All solders are good and I checked them with a microscope
- All IC's and components are from good sources (not from China).
- All electrolytic capacitors are low ESR and high frequency
- No short-circuits

Link to my board: https://github.com/merlinkv/Omega_MSX2

Any idea?

Any help will be much appreciated.

Thanks

skiselev, 2021-12-24:
Well, there's a significant voltage drop somewhere in the system... I am not sure if it just happens because one of the components is faulty, the PCB routing changes, or added peripheral boards.

With the original Omega, similar issue was reported in two cases:

1. When using Rookie Drive or similar with some USB Floppy drives. This is resolved by adding an electrolytic capacitor close to the USB connector
2. When using faulty ICs, the RTC specifically if I recall correctly.

You can try to troubleshoot by measuring the voltage across power pins of different ICs on the system, starting from ones closer to the power connector, and going away from it.

Also you can try using another type of U46 voltage supervisor. Some types, such as ADM693A, have slightly larger tolerance to power supply voltage (minimum of 4.4V instead of 4.65V)

merlinkv, 2021-12-25:
Hi Sergey.

Thanks for your answer.

I have made some changes in tracks and vias.

- Default Track -> Clearance 0.25mm, Track 0.3mm, Via size 0.6mm, Via drill 0.35mm
- GND -> Clearance 0.3mm, Track 0.4mm, Via size 0.7mm, Via drill 0.35mm
- Power (+5v) -> Clearance 0.3mm, Track 1.6mm, Via size 1.6mm, Via drill 0.5mm
- Power2 (+12v y -12v) -> Clearance 0.3mm, Track 1.3mm, Via size 1.3mm, Via drill 0.5mm
- VidAudOut -> Clearance 0.3mm, Track 0.6mm, Via size 0.7mm, Via drill 0.35mm

I you want to see I have uploaded some pictures with the new design: https://github.com/merlinkv/Omega_MSX2

skiselev, 2022-01-04:
A couple comments:
1. Seems to be a loop in the VCC trace on the back layer. Loops in power supply traces is not the best thing.
2. The ground fill patches on top and bottom layers are not well connected. Add more vias to connect between them. This will provide better power supply stability across the board.

merlinkv, 2022-01-14:
Hi Sergey.

First of all, sorry for the delay to answer.

Thanks for your reply. I have mode some changes, now there is not VCC loop and the GND layers have a better connection.

- https://github.com/merlinkv/Omega_MSX2/blob/main/OmegaMB_1_2_5_MKV_VCC.jpg
- https://github.com/merlinkv/Omega_MSX2/blob/main/OmegaMB_1_2_5_MKV_Routed.jpg

Board is tested working with F4-FMPAC & o4Mega (4MB RAM) modules, internal Gotek with LCD 16x2, cartridges Carnivore 2 & TDC-600 (floppy controller) at the same time with one 5.2v/3A PSU.

skiselev, 2022-01-14:
Good to hear it works now!

closed 2022-01-14 by skiselev