skiselev/omega #27: HC vs HCT vs AHCT

alex3331, 2022-01-02:
I noticed that some of the logic IC's are AHCT and some HCT. Is there any reason behind this? I have plenty HC and HCT but will have to buy AHCT (although I understand it is not a lot of money )

skiselev, 2022-01-02:
You can use 74HCT ICs instead of 74AHCT. They are slower, but Omega is slow enough, and the speed difference doesn't matter. I wouldn't recommend using 74HC ICs, since their input levels are not TTL compatible. They will work though, if their inputs are connected to CMOS outputs, e.g., those of 74HCT, 74AHCT, 74HC, 82C55, Z84C00... They might misbehave if the inputs are connected to TTL or NMOS ICs or MSX slots, as they generally use TTL levels.
In the documentation I refer to 74AHCT because they tend to be more available and cheaper.
By the way, you can even use 74LS, or NMOS Z80 and 8255, but they consume more power...

alex3331, 2022-01-02:
Thanks!
Another question I have is regarding keys. I ordered 73 keys instead of 72,
because the space key (higher force) is not available now. Can I make such
a replacement?
Thanks
Alex

skiselev, 2022-01-02:
Yes, you can use regular Cherry MX switch for the space bar. It just would be a bit easier to press...

closed 2022-01-04 by skiselev