skiselev/omega #31: ROM content

alex3331, 2022-01-16:
Where can I find an image for U6 (SST39SF040)?

skiselev, 2022-01-17:
You can use make_roms.sh script here: https://github.com/skiselev/omega/tree/master/BIOSto generate ROM images. It generates several images, I use omega_msx2_msx2+_ntsc.bin on my NTSC system.
The script uses ROMs from original MSX systems. You'll need to find and download them from the internet. I don't post them in there repository for copyright reasons

alex3331, 2022-01-17:
I will find the ROMs, what should I look for?

alex3331, 2022-01-17:
There are plenty of them on the Web. - can you please point me to a website?

skiselev, 2022-01-17:
The NTSC MSX2/MSX2+ images uses the following ROMs:
phc-23jb_basic-bios2.rom
phc-23jb_msx2sub.rom
phc-35j_basic-bios2p.rom
phc-35j_msx2psub.rom
phc-35j_kanjibasic.rom

alex3331, 2022-01-17:
thanks. got these five.

alex3331, 2022-01-18:
I am not able to execute bash command on my Windows computer. Is there an
alternative way of composing these roms into 512 K image?

alex3331, 2022-01-18:
> You'll need a Linux machine to run it (can be a virtual machine too).
> I've sent you the BIOS image in email.

Thank you. I actually ran it on LINUX machine and generated the package
(CBIOS 0.29). However, I can't find assembled 512 k image - just the 16 k
and 32 k pieces. Thanks, anyway.

skiselev, 2022-01-18:
That's weird. Check the script. It expects to find the source ROM images in certain directories/folders. And it doesn't have a really good error handling :)

alex3331, 2022-01-18:
Agree
I could not find a single 512 K image - let me look again at this folder.

elosha, 2022-10-30:
I agree to @alex3331:

After running make_roms.sh with all required scripts, I end up with these files – please note the sizes:
```
-rwx------  1 user  staff  884736 30 Okt 19:43 omega_msx2+_int_ntsc.bin
-rwx------  1 user  staff  901120 30 Okt 19:43 omega_msx2+_ntsc.bin
-rwx------  1 user  staff  950272 30 Okt 19:43 omega_msx2_int_ntsc.bin
-rwx------  1 user  staff  917504 30 Okt 19:43 omega_msx2_msx2+_ntsc.bin
-rwx------  1 user  staff  933888 30 Okt 19:43 omega_msx2_ntsc.bin
-rwx------  1 user  staff  933888 30 Okt 19:43 omega_msx2_pal.bin
```
So none of them fits into my SST39SF020 (256KB)

What am I doing wrong?

Also I noticed @msx-solis' repo here:
https://github.com/msx-solis/OMEGA_MSX_BIOS
His script produces different sizes, still everything is bigger than my Flash ROM. 💇

msx-solis, 2022-11-01:
Hi,
the size of your files should be 512K, the correct size to burn a SST39SF040, if not, you should check each file introduced to the script, if they are not in the route expected or they have an unexpected size the result will fail.
maybe you can do a file for your 39SF020 and maybe it could work but you should check each IC pin to do it.
If you follow that instructions you may understand what does it do and maybe get success:
https://msxmakers.design.blog/proyectos/omega-home-computer/omega-paso-a-paso/cocinando-la-rom-del-omega/

skiselev, 2023-01-06:
@elosha were you able to generate the proper ROM image?

elosha, 2023-01-06:
@skiselev Yes, finally had some spare time again, followed the ROM cooking manual from @msx-solis (easy with school-level spanish) and smacked the ROM parts together.

Hardest part was to understand why there is no MSX2+ PAL ROM, why I don't need one and that it shouldn't distract me. 😅

Actually the included ready-made file "256k_msx2_int_pal_fm.bin" should fit most EU needs I guess.

elosha, 2023-01-08:
Thanks for further clarification 😃 Very helpful. Wasn't kidding, just staying with MSX2 for simplicity reasons (few MSX2+ exclusive titles anyway) and might play with firmware later, after the hardware is fully working. 🤧

closed 2023-06-28 by skiselev