skiselev/omega #32: Is switch 53 illuminated

alex3331, 2022-01-19:
D3 LED is mounted underneath SW53 (CODE/KANA) - what is switch part # for this key?

skiselev, 2022-01-19:
It a usual Cherry MX switch, combined with a **flangeless** 3 mm LED.
The LED part numbers are given in the [Bill of Materials](https://github.com/skiselev/omega/blob/master/Keyboard.md#bill-of-materials---version-10)
It appears that the yellow LEDs (part number VAOL-3HCE4) are no longer available. It can be substituted for red/green/blue/white or any other colors you like

alex3331, 2022-01-19:
I still do not understand. This LED (D3) is right underneath SW53  and the
switch is black and non transparent-  please see attached photo

elosha, 2022-01-19:
@alex3331 You should choose a Switch cap that supports a LED, like some Num Lock keys do.

@skiselev Kingbright (and TT Electronics at Digikey) seem to still manufacture flangeless. I've searched for T1 (3mm) + 3mm outer diameter.

skiselev, 2022-01-20:
@elosha thank you for this information. Do you happen to have the part numbers?

elosha, 2022-04-14:
@skiselev It seems the LEDs I found were not entirely flangeless, but at least barely measurable, adding only 0.5mm outer diameter. In my keyboard I used those and they fit, because the switches have a little vertical gap at the bottom where the LED mounts.

closed 2023-01-06 by skiselev