skiselev/omega #34: No video

alex3331, 2022-01-29:
Turned on the first time; no RGB video on pins 22, 23, 24   of VDP
With the scope:

Synchro  (pins 5, 6) - normal
CPU clock  (pin 8 ) - normal
CPU is listing addresses

V9958 - pins 22,23 and 24 - 2.2 volts, no pulses.
Any way to diagnose?
Thanks

skiselev, 2022-01-29:
It might be not VDP related at all.
The MSX BIOS will not produce any video if RAM or RTC are not working.
I can recommend trying C-BIOS image first, which will produce video output even without working RAM and RTC

alex3331, 2022-01-29:
Thanks, can you give me a reference for the image?

alex3331, 2022-01-30:
I am using RGBS-VGA and RGBS-HDMI  converters (negative CSYNC)  with all my retros; Omega J5 RGB socket does not  seem to have S pin. How is synchronization arranged on J5?
Can I get synchro  from VDP pin 6 (CSYNC)  inverted by  U43C (Q2 installed)? The scope shows correct timing and polarity. I looked at CXA1645 datasheet and it does not mix in SYNC with RGB outputs.

skiselev, 2022-01-30:
I might be mistaken, since I haven't used my Omega with RGB, but I think RGB monitors, at least with SCART, use Composte signal for synchronization

alex3331, 2022-01-30:
You mean Composite Video or Composite Synchro? Composite Synchro is working OK. Will start checking soldering; as well as checking signals with scope.

alex3331, 2022-01-30:
Computer works OK (changed processor) with CBIOS and regular  BIOS. Video
is horrible. It does not want to talk to my adapters. I need to figure out
how to connect it to LCD monitor. Both my adapters so far do not work with
OMEGA RGB output. How do people connect it?
Alex

skiselev, 2022-01-31:
I personally use S-Video. Composite video also works for me, although with lower quality.
Several people reported success with using RGB monitors or TVs with SCART connector (common in Europe). In the latter case, the connector provides RGB and composite video, and the TV/monitor extracts synchronization information from the composite video signal.
There are a few, not particularly expensive S-Video to HDMI converters on Amazon, such as this one: https://www.amazon.com/Converter-Composite-Adapter-Support-Blue-Ray/dp/B07GVDWGK8/
Maybe give it a try? (and return of it doesn't work)

alex3331, 2022-01-31:
I sure will. Let me play with the scope a bit more. I have seen somebody
taking RGB from 9929 and adding a 2N2907 transistor amplifier to each
channel to bring it to TTL level. I have the PCB already so may try.

skiselev, 2022-01-31:
Analog RGB voltage is normally much lower than 5V. For example, VGA ranges from 0V (black/no color component) to 0.7V (white/full intensity).
CGA and EGA used 5V TTL digital signal, but I assume that's not what you want.

cristianoag, 2022-02-28:
I'm using mine with RGB currently, but I noticed it is super sensible to the power source. I have multiple 5V power supplies, including my bench device and I only manage to make RGB work with one of the power supplies. I didn't check with scope yet, but that is strange. While it works ok in S-Video and Composite with any 5V power source, RGB only works within specific parameters.

alex3331, 2022-02-28:
After I fixed my VDP issue RGB works fine for me. I use cheap RGBtoVGA
adapter based on 8220 chip. The board is $25 on Amazon. Power supply is a
cheap 5 v 2.5 a. If everything is OK, Omega does not pull a lot of current

cristianoag, 2022-02-28:
What was the issue with your VDP?

alex3331, 2022-02-28:
As it turned out, VDP was fine ; two RAM pins were bent, so VDP did not
initialize properly. Now I have an extra chip which is tested and working.

closed 2023-01-06 by skiselev