skiselev/omega #35: Hardware switch & HDMI

EMLommers, 2022-03-10:
Hi,

This is the best successor of MSX, I have seen!
No emulation, but real Zilog inside, much better than FPGA, ARM or whatsever.
If there is a possibility to have a hardware switch to select MSX 1/2/2+/tR, (incl. button R800 on/off, maybe for the suckers like me an additional switch for 7Mhz) with flashable ROMs for each type and a HDMI output,
there is no possibility to deny this... and really a wanna have!..

I know it is not easy.. but possible!

skiselev, 2022-03-10:
Thanks for your feedback!

A few thoughts:

- Switching between MSX1/2/2+ is practically impossible. The defining difference between MSX1, MSX2, and MSX2+ is the VDP type - TMS9918, V9938, and V9958 respectively. To make a switch, the board would have to have all three VDPs and somehow switch between them. There are other, less significant differences too - BIOS/BASIC (there is an option to load two BIOS images right now, and easily switch between them - that might work as switch between MSX2 and MSX2+ BIOSes), memory mapper in MSX2 (but also present in some MSX1), F4 port in MSX 2+ (no one seems to care about), but these tend to be backward compatible
- Turbo R uses R800 CPU. Good luck finding one, and also soldering SMD components is a problem for many people
- Turbo mode (7.159 MHz) on Z80 CPU might be possible. It is on my list of possible enhancements for the next version
- HDMI output is likely no go, unless there is a reasonably small and affordable RGB to HDMI converter module that can be installed on the PCB.

EMLommers, 2022-03-15:
Hi Skiselev,

Thanks for your reply.
If there is an possilibity to use VGA to HDMI adapter (external) , would be  appreciated, due to currrent monitors, use vga /hdmi (some times scart)
I am used to run MSX2, with switch to 2+ and switch for 7Mhz. Running an MSX2 and able to access the VDP registers of V9958, would be excellent. 7Mhz is for me only used to check timings adressing VDP.
One last question,  possible to connect my old MSX drive to the board (or 2 like Phlips NMS8280 ??) To run original disk software.

What would be the estimated price when finished?

skiselev, 2023-01-06:
@EMLommers apologies for a very late reply

* MSX Turbo-R compatibility is out of the scope. As far as I can tell, it is not easy to implement it. Z380 does not seem to be a straight forward replacement for R800
* Some people do use 3-rd party RGB to HDMI adapters. Currently there are no plans to integrate one in the system. Also, supposedly, some monitors can use the RGB signal with NTSC timing directly if connected to VGA input
* I still don't understand switching between MSX2 and MSX2+... In Omega you can have both MSX2 and MSX2+ ROM images, and switch between two. I think V9958 is backward compatible with V9938, so that should work just fine. As I said previously, 7.159 MHz is on the list of possible enhancements for the next version
* Currently Omega does not have a built-in floppy disk controller. With that being said, it is on the list of possible enhancements for the next version. Also you can use a floppy disk controller cartridge, for example, FDC-600, [Rookie Drive](http://rookiedrive.com/en/home/), together with whatever floppy drives they support
* At this point I am not planning to remove printer port
* I cannot give you a price estimate. It depends where you get the parts, what case you use, and so on...

closed 2023-01-06 by skiselev