skiselev/omega #36: Is there a way to run 128 K or larger rom files on Omega

alex3331, 2022-03-27:
It looks like slot address bus can address no more than 64 k. Is there a way to run larger programs without the need of another external storage device, like FD, SD card or hard drive?

skiselev, 2022-03-27:
The address bus is 16-bit, so the addressable memory is limited to 64 KB.
There are ways around this limitation, usually by "paging" the memory. MSX2 memory mapper, and MegaROMs are the examples of such approaches.

alex3331, 2022-03-27:
Can you please give  a reference to MegaROM and MSX2 memory mapper? I have a few 128 k, 256 k and 512 k programs and would like to try them

skiselev, 2023-01-06:
@alex3331 I believe I've provided you the information in the email.
Closing the issue

closed 2023-01-06 by skiselev