skiselev/omega #37: HC vs HCT - BOM / Silkscreen mistmatch

rain2usa, 2022-06-18:
Good Evening.  I'm having a lot of trouble building the Omega.  I bought the quick kit from eBay with all LS chips but that didn't work so I bought everything as-is from the BOM as much as possible.  Anyways, I just read up on the differences between HC vs LS vs HCT stating HC isn't TTL compatible and shouldn't be used.  One of the BOM parts is asking for a regular HC IC.  Is this correct?  

Integrated Circuit | U15 - U18 | 4-by-4 Register File - 74HC670, 16 pin DIP.  

Silkscreen is asking for 74HCT670.  Could the difference be causing my "Error: Memory not found" screen no matter what IC I keep/pull?  GALs and Flash are programmed correctly.  Voltages are all good. The Smoke tests 1 - 3 from the MSX Makers guide seems to work. The Expected scoped frequencies in many places seems to match, but not the PPI_CS.  Can't get any frequency reading from the 82c55 (Pin 6) to the U12 (Chip Select Pin 14).  I get 4.7V on Pin 6 (U2), which I have been told may not be correct.
I listed all of my issues on the MSX Makers forum if it helps any to see what all I have done.  Any help would be appreciated.

skiselev, 2022-06-18:
74HC inputs are not compatible with 74LS outputs.
74HC inputs are compatible with 74HCT and other CMOS (e.g. GALs, Z84C00 CPU, 82C55 PPI) outputs.
Look at the schematic and check if you have any 74LS outputs going to 74HC inputs
I'll check the schematic later regarding your measurements

rain2usa, 2022-06-18:
Thank you for the quick reply. Ok, then the 670 is fine then.  All of the ICs are HC/HCT/ACHT ICs except the lone LS07.  The only HC IC is the 670 in my current configuration.  I'll wait for your review of the measurements and appreciate the help.  I just feel like I'm missing something and no one on the Makers Forum knows what that could be.  Just to reiterate from the Maker's Forum, I have a 1.4 revision board.

rain2usa, 2022-06-22:
Defective/Incompatible Mitsubishi 82c55 was the cause of the board issues.  Replaced with exact part number from BOM and it's working thus far.  Was unable to diagnose Why it was defective but it was also the last part in that part of the circuit that wasn't replaced until a couple days ago.

closed 2022-06-22 by rain2usa