skiselev/omega #39: Make Ys available on the RGB connector

sdsnatcher, 2022-07-03:
Please add the Ys signal to the RGB connector. The pin-5 of the DIN-8 MSX RGB connector was meant to be the Ys signal from the VDP.

To save some pennies on a transistor, many MSX models came with just a resistor divider connected to that pin.

But for those on the SCART world, it's very useful to have a funcional Ys pin, since the TV can superimpose the MSX image over other sources.

I attached here a screenshot of how it was implemented on the Philips NMS-8245.

[image]

sdsnatcher, 2022-07-03:
There are a lot of free inverters on the current Omega schematic, and any of them could also be used to implement the Ys signal too:

[image]

skiselev, 2022-07-03:
Thanks for the suggestion! I'll add it to the next revision.

Leshak-pcb, 2023-06-29:
> There are a lot of free inverters on the current Omega schematic, and any of them could also be used to implement the Ys signal too:
> 
> [image]

Since Ys is terminated with 75 Omh resistance this is probably too much for 74HCT04. Increasing series to 150 Omh results in 18.7 mA current - pretty high, but acceptable for 74HCT.
74ACT04 can source up to 24 mA, but is it suitable for a clock generator?