skiselev/omega #41: Any reason forJP7 and RR10 to exist?

sdsnatcher, 2022-07-21:
Is there any reason for JP7 and RR10 to exist? They only seem to be a source of trouble for those that I'm seeing to assemble the project.

Some of them cut the JP7 tracks open, causing mirroring of the 512KB of RAM to happen amongst the Memory Mapper pages. This is an unwanted behavior that causes some programs to incorrectly detect up to 4096KB of RAM from the 512KB RAM chip.

skiselev, 2023-01-05:
Yes, there is a reason. The previous revisions (1.1, 1.2) didn't have JP7, JP8, and RR10. This resulted in some poorly written software (Jordi/MSXMaker knows exactly which) to misdetect the memory size. If I recall correctly, the software relied on writing and reading back page register, which supposedly should wrap around when the top of the memory is reached. In case previous revisions, the page register has 8-bit, while only 5 bits are used on a 512 KB system. So we added these jumpers.
Perhaps I should document them better...

skiselev, 2023-01-05:
Some other software, detected the memory size by writing values to all 256 theoretical memory pages, and then reading the written values back, expecting that the values will be overwritten at some point, if the memory size is less than 4 MB. In case of the earlier Omega design, the writes above 512 KB (32 pages) simply disappear, which was causing issues.

skiselev, 2023-01-05:
[image]
This shows the "testmap.com" output that displays an error, before implementing the "fix". Generally speaking, the software shouldn't read the page register back and rely on its value... but it does...

closed 2023-06-28 by skiselev