skiselev/omega #42: Which Programmer to buy?

CaffeinatedTech, 2022-10-27:
I can't seem to get hold of a minipro TL866II any more.  They are either out of stock, or quite expensive.

Can anyone suggest a currently available programmer which would be suitable for this project?

I've got an arduino mega if anyone can point me to a project to build a programmer.  I'm happy with that option too.

skiselev, 2022-10-27:
I think TL866II Plus and Xgecu are the later revisions of TL866 series
Check the programmers' device list for SST39SF040 and ATF16V8 support

CaffeinatedTech, 2022-10-28:
Looks like TL866II Plus was replaced by the XGecu T48, and the T56.  The T56 is more capable and faster, but is considerably more expensive.  So I'll try to get a T48.

Here's the [device list](http://www.xgecu.com/MiniPro/T48_List.txt) for anyone interested.  Both of the specified chips for this project are listed.