skiselev/omega #43: Proposed box of various designs for MSX Omega

Mundo-Desconocido, 2023-01-08:
I have some conceptual proposals on 21st century case designs for the MSX Omega.

As a background, I helped the MEGA65 team for a year on some modifications to the computer case as well as some conceptual ideas for future versions.

I would be grateful if you could rate the different ideas I uploaded in the following thread.

Muchas gracias
Jose Luis
Mundo Desconocido
[image]

skiselev, 2023-01-12:
Jose Luis,

These designs look beautiful.
A few things of concern:
* I think the design assumes that the keyboard is housed separately from the computer. With the current computer design, it would require a 16-wire cable to communicate between the keyboard and the computer. It is possible that systems with separate keyboard design used a different approach that allowed reducing the number of wires. I'll have to check how easy that can be implemented
* Currently the slot connectors are stacked in the right corner of the motherboard. Moving them elsewhere would require redesigning the motherboard, or developing an extra board with the slot connectors. The particular "side by side" arrangement might be more problematic, as it will make more difficult to route the PCB traces
* Looks like the design includes places for floppy disk drives. I like that, although we need to come up with a way to connect them to the computer. Currently it would require using a controller in one of the cartridge slots. Adding a floppy disk controller is on my enhancements list. I don't quite understand what SSD is and why there is a bump in the case to accommodate it. Is it supposed to be a removable drive? I think most if not all SSDs have SATA interface, so it will be difficult to connect them to an MSX. It is more reasonable to connect an SD card or a USB-based storage devices
* I personally do not recommend using MSX anywhere in the labeling, as it is a registered trademark 

Also, what is the goal of this case design? Are you planning to manufacture it eventually? Or would you publish the design files, e.g., 3D printable files (STL or similar)?

Thanks,
Sergey

Mundo-Desconocido, 2023-01-14:
> 

Thank you very much for your indications and assessment.

This is just one of the conceptual ideas I have for the Omega, obviously, I will upload more in this thread as I have them designed, another one is about to come out of the oven.

Regarding the non-inclusion of the MSX logo, I will take it into account, it happened the same with MEGA, etc...

The purpose of the case that I design here, is to offer it for free to the community in a printable STL file or if this project wishes, to use it as a final capsule in injected plastic, if someone decides to market them, for my part, my only interest is the mere altruistic collaboration.

thanks
JL

Mundo-Desconocido, 2023-01-14:
This is a new concept.

It has shapes reminiscent of stealth forms and the colours are warm and classic, a strange mix that together with the blue and red stripes give it a modern and classic touch at the same time.

Remember that these are just sketches. Whichever one you choose, it will fit the shape of the motherboard.

Thanks for your feedback.
JL
[image]
[image]
[image]

nullvalue0, 2023-01-22:
I really like the idea of a separate keyboard and was thinking about doing my build like that. Since it requires a 16-pin cable, I was planning using an [ODB2 cable like this](https://www.ebay.com/itm/291181598660).

nullvalue0, 2023-01-25:
@Mundo-Desconocido I already have an enclosure that I am planning to use for my build for the motherboard but I would really love to get the STLs for the keyboard case in your first design. Not sure how much work that would be..

skiselev, 2023-01-27:
@Mundo-Desconocido that's an interesting concept and lovely colors! Is it possible to do the floppy part a separate module, pluggable into the computer? There might be some mechanical considerations there, as it would need to be secured, so it doesn't accidentally disconnects, or worse, breaks the connector

sallevan, 2023-03-21:
Sergey,

> I think the design assumes that the keyboard is housed separately from the computer. With the current computer design, it would require a 16-wire cable to communicate between the keyboard and the [...]

I have nice times in the school which has "KUVT "Yamaha" (КУВТ - комплект учебной вычислительной техники), where schoolchildren's computers were "all-in-a-keyboard" design (without FDDs) while teacher's one has main unit (with FDDs) **separate from keyboard.**
As such, I think that Yamaha, Inc. somehow resolved the issue with a bunch of keyboard wires those years. Unfortunately, I have no design/schematic files for such a variant. :(