skiselev/omega #45: spacebar stabilizer part no is wrong

mrkite, 2023-02-10:
In the parts list for the keyboard, you list Mouser 540-G99-0226 for a "Leveling kit for MX 1x8".  That is the right description and part for a 1x8 stabilizer, but the spacebar used in the project is a 6.25u spacebar.

skiselev, 2023-02-10:
I know. Unfortunately, Mouser does not have a stabilizer for 6.25u spacebar. 
I used a longer 8u stabilizer, bent and cut it to build a shorter one.
Perhaps this needs to be documented 😄

nullvalue0, 2023-02-11:
I just ordered them from here: https://www.aliexpress.us/item/2251832533420646.html

closed 2023-03-10 by skiselev

skiselev, 2023-03-10:
I've enquired Cherry US about ordering these parts. The part name and the number are: Wire yoke 6145017
They said that there is no official distributor that sells these parts.
Just in case, I'll shoot an email to Digi-Key and Mouser, and ask them if they want to stock that.

nullvalue0, 2023-03-10:
I can confirm the part I listed above from AliExpress fit perfectly for me.
I got the 7U 60 87 version. I know it might not sound right but it fit
without modification.