skiselev/omega #47: Keyboard LEDs

peterw8102, 2023-03-10:
The keyboard has D2 and D3 as 3mm LEDs (through hole 3mm from the BOM). I can't see how these work with the MX Cherry Switches in the BOM. There's no room for the LED. I'm assuming I'm missing something? I thought I'd bought the right switches. Any help appreciated.

skiselev, 2023-03-10:
Hope this explains :)
You'll want to get a flangeless 3 mm LEDs. See the BOM for some suggested part numbers. 
If for some reason your keys don't have a place to mount LED, perhaps you don't have the original Cherry MX switches?
[image]

closed 2023-03-10 by skiselev

peterw8102, 2023-03-10:
Brilliant, I do have the right switches and had completely missed those holes! Thank you!