skiselev/omega #49: Floppy disk interface TDC-600 clone Error.

drawingangnam, 2023-06-02:
I successfully assembled a floppy disk interface TDC-600 clone. 
I connected to OMEGA and connected GoteK flash floppy to run the game. 
However, while reading the file, it freezes on a black screen. (But interestingly, some games are running well.) 
Gotek Floppy Action LED is on but OMEGA is stopped. 
Some games have a start screen, but will stop after a while. Is it an MY Omega problem? 
Or is it a TDC-600 clone issue? 
The TDC-600 clone was tested by assembling two different production years WD37C65.

[image]
My  TDC-600 clone

[image]
"Fire Hawk"
logo comes out and reading the disc, and at the bottom of the page, there's a broken graphic like that and it stops.

merlinkv, 2023-06-03:
Hello,

As I can see, you only have the gotek connected. Try this:

- Set all jumpers UP
- Plain cable, set the Gotek as DS0
- Standard PC FDD cable: Connect the Gotek after the crossing cables and set it as DS1

drawingangnam, 2023-06-03:
> Hello,
> 
> As I can see, you only have the gotek connected. Try this:
> 
> * Set all jumpers UP
> * Plain cable, set the Gotek as DS0
> * Standard PC FDD cable: Connect the Gotek after the crossing cables and set it as DS1
Thank you for your interest. I've already tested everything you said.

I've done everything I can think of.

Jumper setting, cable related, firmware related…

It's still the same symptom.

I asked the community to test the person who made OMEGA before me.

I heard that there was a same problem when I used the all-in-one GoTek cartridge using WD37C65 PLCC-44.
[image]

He said THAT worked well at SONY MSX.

merlinkv, 2023-06-03:
I don't know ... but I have two TDC-600 clones (https://msxmakers.design.blog/proyectos/proyecto-tdc-600/) working like a charm with Gotek and real FDD's.

But I think is not a problem of the Omega or the TDC-600 controller. I have been tested this game (Fire Hawk) and have the same problem as you.

Maybe is a problem of bad cracked/dumped game? Becasue many MSX1/MSX2 games works perfectly.

drawingangnam, 2023-06-03:
> 잘 모르겠지만 ... Gotek 및 실제 FDD와 함께 매력처럼 작동하는 두 개의 TDC-600 클론 ( [https://msxmakers.design.blog/proyectos/proyecto-tdc-600/ )이 있습니다.](https://msxmakers.design.blog/proyectos/proyecto-tdc-600/)
> 
> 하지만 오메가나 TDC-600 컨트롤러의 문제는 아니라고 생각합니다(제가 틀렸을 수도 있습니다).
> 
> 나는 이 게임(Fire Hawk)을 시도했고 당신과 같은 문제가 있습니다. 그러나 대부분의 MSX1/MSX2 게임(dsk)은 완벽하게 작동합니다.

> I don't know ... but I have two TDC-600 clones (https://msxmakers.design.blog/proyectos/proyecto-tdc-600/) working like a charm with Gotek and real FDD's.
> 
> But I think is not a problem of the Omega or the TDC-600 controller (maybe I'm wrong).
> 
> I have tried this game (Fire Hawk) and I have the same problem as you. But most of MSX1/MSX2 games (dsk) work perfectl

> I don't know ... but I have two TDC-600 clones (https://msxmakers.design.blog/proyectos/proyecto-tdc-600/) working like a charm with Gotek and real FDD's.
> 
> But I think is not a problem of the Omega or the TDC-600 controller (maybe I'm wrong).
> 
> I have tried this game (Fire Hawk) and I have the same problem as you. But most of MSX1/MSX2 games (dsk) work perfectly.

Thank you for the test.

It runs well on my DAEWOO MSX with GOTEK that doesn't run on OMEGA. (Of course, the FDC is the DAEWOO MSX built-in interface.)

If possible, could you test it for me?

Psycho World
Ultima4
Snatcher
SD Snatcher
Ancient Ys Vanished II - The Final Chapter
Valis II - The Fantasm Soldier

merlinkv, 2023-06-03:
I have done some test with Gotek and real disks.

Psycho World -> NO
Ultima4 -> OK
Snatcher -> OK
SD Snatcher -> I can't test, I have SCC (Carnivore2) but ask for Sound Cartridge ... ??
Ancient Ys Vanished II - The Final Chapter -> NO
Valis II - The Fantasm Soldier -> OK

drawingangnam, 2023-06-03:
> I have done some test with Gotek and real disks.
> 
> Psycho World -> NO Ultima4 -> OK Snatcher -> OK SD Snatcher -> I can't test, I have SCC (Carnivore2) but ask for Sound Cartridge ... ?? Ancient Ys Vanished II - The Final Chapter -> NO Valis II - The Fantasm Soldier -> OK

Thank you very much for the test.

But it's interesting.
These are all games that don't run on my OMEGA but some games run on your OMEGA!!

I'm really curious.
If it's caused by a mechanical defect, the same reaction should come out..

merlinkv, 2023-06-05:
Ancient Ys Vanished II - The Final Chapter (1988)(Falcom)(Disk 1 of 2)(Game Disk)[tr es Manuel Pazos].dsk
Ancient Ys Vanished II - The Final Chapter (1988)(Falcom)(Disk 2 of 2)(Game Disk)[tr es Manuel Pazos].dsk

Works fine.

drawingangnam, 2023-06-05:
> Psycho World [T-En by FRS v.1.0].dsk

Thank you for the TEST.
I downloaded the exact same file and tested it.

Ancient Ys Vanished II - The Final Chapter : Stop as soon as the step roll FIRTST LINE comes out

Psycho World [T-En by FRS v.1.0].dsk : START selected and the "LOADING" appears when the game is pressed

There is a high probability that my OMEGA has a problem.

Did you use a chip with the correct prefix?

AHCT138 -> F138
HCT670 -> HC670
HCT153 -> HC153

Except for this replacement, we used the exact chip.

Can this cause the problem?

merlinkv, 2023-06-05:
I'm not sure, but yes, that may be the problem.

I use chips with the exact nomenclature and with my own version of the motherboard (It's almost the same as the original 1.4EU, except that you can choose between 4 ROMs and it has some extra power connectors).

https://github.com/merlinkv/Omega_MSX2

drawingangnam, 2023-06-06:
> I'm not sure, but yes, that may be the problem.
> 
> I use chips with the exact nomenclature and with my own version of the motherboard (It's almost the same as the original 1.4EU, except that you can choose between 4 ROMs and it has some extra power connectors).
> 
> https://github.com/merlinkv/Omega_MSX2

I ordered new HCT chips.

I will replace the HC with HCT and report the results.

And 

I'm thinking about build a new FDC that requires 2793.
(https://github.com/konkotgit/fdd-cdx2)

merlinkv, 2023-06-06:
Great!

I'll wait your tests.

drawingangnam, 2023-07-05:
> > Psycho World [T-En by FRS v.1.0].dsk
> 
> Thank you for the TEST. I downloaded the exact same file and tested it.
> 
> Ancient Ys Vanished II - The Final Chapter : After the first line of the step roll comes out, stop with the Gottek ACT LED illuminated.
> 
> Psycho World [T-En by FRS v.1.0].dsk : When the intro comes out and the START menu is selected, "LOADING" is displayed, but the ACT LED is on and stops.
> 
> There is a high probability that my OMEGA has a problem.
> 
> Did you use a chip with the correct prefix?
> 
> AHCT138 -> F138 HCT670 -> HC670 HCT153 -> HC153
> 
> Except for this replacement, I used the exact chip.
> 
> Can this cause the problem?

HC670 -> HCT670
HCT245 -> F245
HC153 -> LS153 (Ori. HCT153)

I've tried this.

The problem has not been resolved.  ;'-(

FDD is freezing with the ACT LED on while loading the game.