skiselev/omega #52: Way better s-video picture quality than rgb

Sch-LikA, 2024-01-10:
Hi All !

First a huge thank to the MSX Omega community, specially Skiselev, I built a nicely featured Omega with fm-pac, 4.5Mb memory, a full set of cartridges, all open-source ! I learned a lot in the process !

Now a strange issue. My RGB output (to a scart cable) is noticably uglier than my s-video output. The rgb is washed, noisy where the s-video is extremely clean. Since they go the same route through the same IC's I don't get it. Is the composite video as composite sync a possible cause ? It may be as I noted that if I leave the composite video plugged in the rgb output is even worse. Anybody tried using sync from some other place, like directly from the 9958 ?

Thanks !!

skiselev, 2024-01-10:
Hi,

Unfortunately I don't have an RGB monitor, so RGB functionality is not well tested.
It was recommended to use S-Video Luma signal instead of the composite for the CSYNC signal:
https://github.com/skiselev/omega/issues/38
It would be great if you can test this approach.

Thanks,
Sergey

Sch-LikA, 2024-03-23:
Hi !

I confirm, S-Video Luma instead of Composite for CSYNC gives much better results. CRT monitors like Commodore 1084S do not care much, but more modern stuff, including GBS 8200 give much cleaner results.

Thanks,
Marcel

closed 2024-03-23 by Sch-LikA