skiselev/omega #56: Joystick issue: JB1 should be connected to TRG2_A but is connected to TRG1_B

Pax-nl, 2025-04-26:
Hi,

I found an issue with the Joystick port, pin 11 and 13 on U29 should be reversed.

[image][image]

I first encountered the issue with Super Sokuban, which is using MSXgl and Aoineko and Ludo from the MSXgl discord found the cause. 

Both me and Ludo debugged the issue by switching pin 11 and 13 on U29/74HCT157

My test setup:
[image]
And Ludo with a nice cnc pcb:

[image]

[image]

herraa1, 2025-04-27:
The actual fix seems to be swapping signals for pin 11 and 13 of U42/74LS07 (not pins 11 and 13 of U29/74HCT157).

See [https://www.msx.org/forum/msx-talk/hardware/msxmakers-support?page=17](url) for additional information.

Pax-nl, 2025-04-27:
> The actual fix seems to be swapping signals for pin 11 and 13 of U42/74LS07

Sorry my mistake, updated..

herraa1, 2025-04-27:
> Sorry my mistake, updated..

Np. 

> The post of Ahmsx is very detailed and explains the issue perfectly..

Thanks, it’s me.

skiselev, 2025-04-28:
Thanks for analyzing and reporting the issue. I am planning to fix the schematic and update the PCB in the near future. Your PCB, or perhaps just lifting U42/74LS07 pins 11 and 13 and using a couple of pieces of wire to connect them to the pads corresponding to pins 13 and 11 respectively can be used as a workaround for existing systems.

skiselev, 2025-04-28:
Comment for myself: The joystick implementation is described in the [MSX Technical Data Book](https://map.grauw.nl/resources/system/msxtech.pdf), page 26 (page 29 in the PDF)

skiselev, 2025-05-01:
Fixed in version 1.5. Described a workaround/patch for previous versions.

Pax-nl, 2025-05-02:
@skiselev thank you for the quick solution.

herraa1, 2025-10-04:
For existing 1.4 mainboards, this [omega-1.4-joyfix-board-v1 interposer board](https://github.com/herraa1/omega-1.4-joyfix-board-v1) which fixes the joystick circuit bug may be useful.