skiselev/omega #57: Schematic/PCB issue with JB1 and JB2 on PSG

gfx0101, 2025-04-27:
We identified an error in the schematics and the PCB of the Omega, the JB1 and JB2 are wrongly connected to the PSG and should be swapped.

[image]

[image]

[image]

Without this modification the output pin 6 of the port A is driving the pin 7 of port B and the output pin 7 of port B is driving the pin 6 of port A.

See schematics of a standard MSX:
[image]

BTW i love the Omega, thanks for this great project.

herraa1, 2025-04-27:
Duplicate of #56

gfx0101, 2025-04-27:
Ah yes sorry I just seen it, at least you are well aware now :)

closed 2025-04-28 by skiselev