skiselev/omega #58: keyboard internationalization option

mpsantos7, 2025-04-27:
Would it be possible to make an optional keyboard with support for dead keys?  I believe that the base could be some of the MSX from Philips, but there are also Brazilian and Argentine models.  Although I don't know how the support for this would be in the case of MSX 2+ BIOS  ...

skiselev, 2025-04-28:
Sorry, I don't have plans to implement any optional keyboard support.
There are multiple reasons for this:
* I personally don't know what keyboard mapping, matrixes, key locations are on the localized MSX systems. If I would want to do that, it will take a lot of my time to research this.
* Different layouts would likely mean implementing different keyboard PCBs. Shouldn't be too difficult, but again I have no time for this.
* I don't think MSX2+ ever supported anything but Japanese layout. It might be possible to patch it to support other keyboards layouts, but again, beyond my level of expertise.

With that said, this project is open source, so feel free to clone the GitHub repo, add support for whatever keyboard layout you want, and contribute back for everyone's enjoyment.

Nephrite-FM, 2025-05-07:
I have a Question on this, too.  You have on the mainboard options a jumper setting (JP2) to set it as a JIS layout.

The Standard ANSI layout, has no katakana ロ (Ro) key.  Is that what the DEAD Key switch footprint is for?  So it can handle this extra, quite necessary to JIS-layout ロ key?  I can't tell yet as your Circuit diags/silkscreens don't mention the kana, only the english ANSI layout, with no mention as to what the Motherboard would accept (i.e. what Switch Rows and Columns correspond to what in the MSX2+ Kana keyboard), and I haven't finished building the mainboard yet (Need to order the rest of the parts) so I can't plug it in and test it even.

I also noticed this:

https://github.com/skiselev/omega/issues/46

is the newest Keyboard layout work with ロ in the dead key space, and no @ key?